conneroisu/processsor-proj-1 #39: Implement your new instruction fetch logic using VHDL. Use Modelsim to test your design thoroughly to make sure it is working as expected. Describe how the execution of the control flow possibilities corresponds to the Modelsim waveforms in your writeup.

conneroisu, 2024-10-24:


conneroisu, 2024-10-29:
addi 01:
[image]

conneroisu, 2024-10-29:
add02:
[image]

conneroisu, 2024-10-29:
addi 1:
[image]

conneroisu, 2024-10-29:
addiu 1:
[image]

conneroisu, 2024-10-29:
addu 1:
[image]

conneroisu, 2024-10-29:
and 1:
[image]

conneroisu, 2024-10-29:
j1:
[image]

conneroisu, 2024-10-29:
jr1:
[image]

conneroisu, 2024-10-29:
jal 1:
[image]

aidanfoss, 2024-10-29:
the goat

closed 2024-10-30 by conneroisu