conneroisu/processsor-proj-1 #46: In your writeup, briefly describe your design approach, including any resources you used to choose or implement the design. Include at least one design decision you had to make.

conneroisu, 2024-10-24:
Talk about how we used github issues to track all need-to-do tasks

aidanfoss, 2024-10-25:
i write this portion @conneroisu, but i'm not very happy with it. I wrote "A major design decision we had to make was based around the barrel shifter. After a few drafts, as well as talking to other students, a solution involving an extra mux per stage to determine logical vs arithmetic shifts was discovered, as well as flipping the bits to do left vs right shifting."

aidanfoss, 2024-10-25:
> Talk about how we used github issues to track all need-to-do tasks

We chose to use github to help implement our design. By assigning every task an issue, we could easily work asynchronously and collaborate through the comments on the repository. This allowed us to create the issues in bulk, and then self assign and rearrange tasks if need be. By opening up each other’s branches, we could view exactly how our team members had their components designed, which allowed for easier collaboration and troubleshooting.

closed 2024-10-25 by conneroisu