conneroisu/processsor-proj-1 #69: rebase awfoss to aidanfoss

aidanfoss, 2024-10-25:
:(

aidanfoss, 2024-10-25:
$ git filter-branch --env-filter '
WRONG_EMAIL="wrong@example.com"
NEW_NAME="New Name Value"
NEW_EMAIL="correct@example.com"

if [ "$GIT_COMMITTER_EMAIL" = "$WRONG_EMAIL" ]
then
    export GIT_COMMITTER_NAME="$NEW_NAME"
    export GIT_COMMITTER_EMAIL="$NEW_EMAIL"
fi
if [ "$GIT_AUTHOR_EMAIL" = "$WRONG_EMAIL" ]
then
    export GIT_AUTHOR_NAME="$NEW_NAME"
    export GIT_AUTHOR_EMAIL="$NEW_EMAIL"
fi
' --tag-name-filter cat -- --branches --tags

aidanfoss, 2024-10-25:
@conneroisu

conneroisu, 2024-10-25:
What's up with this?

aidanfoss, 2024-10-25:
I incorrectly logged in at first my commits are credited towards my ISU account which doesnt actually exist on github. That command would edit the author of all my commits to apply it correctly.